Open-CMSIS-Pack/generator-bridge #95: Keil multiply definitions when multi-target

minicarpet, 2024-11-26:
**Describe the bug**
Using Keil uVision with project having 2 multi targets, we need to generate CubeMX file for each target, then all files is used from project which occurs to multiply definitions of all generated functions.
Maybe this is missunderstood how to use it.

**To Reproduce**

1. Starts Keil
2. Create a project
3. Add an other target
[image]
5. Generate for each target CubeMX files as asked
[image]
[image]
6. Compile
[image]

**Expected behavior**

- All targets should use the same generated files OR
- Include only generated files for the target (exclude all others)

**Environment (please complete the following information):**

- keil uVision: 5.41
- STM32CubeMX 6.12.1
- cbridge: generator-bridge version 0.9.11 (C) 2023-2024 Arm Ltd.
- device: STM32L4P5CEUx, packs: ARM::CMSIS 6.1.0 ; Keil::STM32L4xx_DFP 3.0.0
- OS: windows

**Additional context**
I used as workaround:
- modify path of CubeMX generator into `C:\Keil_v5\ARM\cmsis-toolbox\etc\global.generator.yml`:
```
generator:
  - id: CubeMX
    description: Global Registered Generator
    download-url: https://www.st.com/en/development-tools/stm32cubemx.html#st-get-software
    run: ../bin/cbridge
    path: $SolutionDir()$/STM32CubeMX
```
- Adapt cbridge go code (see [my commit](https://github.com/minicarpet/generator-bridge/commit/c99fa45723a94673267a44b544fe9fdfabcf100d) from forked repository)
**Note**: The modification of `internal/stm32CubeMX/mxDevice.go` is done because I use peripheral generation as pair

**OR**

What I can do manually is to not include generated files dependings on target:
[image]

jkrech, 2024-11-26:
Thanks for reporting this bug. 
The default path configured in the CMSIS-Toolbox:
```
    path: $SolutionDir()$/STM32CubeMX/$TargetType$
```
separates the generator based on "TargetType". In the CMSIS Solution project format there is target-type and build-type. Target type must specify the used device and therefore two target-types are expected to use separate generator configurations.

uVision will therefore expect two independent STM32CubeMX projects for two uVision "targets".
The bug in uVision seems to be that the loading of the generator information is not correctly separated between the targets.
This needs investigation.
As far as I can see this problem should be fixed in MDK uVision.

minicarpet, 2024-11-26:
Well in that case, this is done through cgen.yml generation (through `WriteCgenYml`)
Groups should maybe renamed ?

edriouk, 2025-01-15:
Problem in uVision is Fixed and will appear in the next MDK Release (~ February 2025)

jkrech, 2025-04-01:
MDK 5.42 got released. Can this issue be closed?

minicarpet, 2025-04-04:
This is better and I can compile now:
but there is multiply definition always print:

[image]

this not a real issue but this is 'ugly'

jkrech, 2025-04-14:
@minicarpet can you please check whether each project target uses a separate directory for the object files?

See "Options for target" -> Tab: Output -> Button: "Select Folders for Objects..."

jkrech, 2025-04-15:
For uVision Keil MDK product support please contact [Arm support](https://services.arm.com/support/s/contactsupport).

closed 2025-04-15 by jkrech